kmpm/nodemcu-uploader #11: Add ability to view log

pastukhov, 2015-04-01:
Please add ability to attach nodemcu console to view logs.

lrozema, 2015-04-01:
Can you collaborate on what you mean by this?

To give some insight in what I am currently cooking up (maybe it is what you are looking for). Currently nodemcu-uploader does upload and verify. I am extending this with three options:
* An option to execute dofile() on the newly uploaded file, maybe with an additional reset in between.
* Next an option to start miniterm right after upload (or dofile)
* Finally an option to monitor for file changes on the selected files and re-upload them.

This way you can keep vim (I assume this is everyone's favorite editor) open and just hit save, the running instance of nodemcu will then automatically upload the new file, dofile it and have you back in miniterm to view what is happening. Kind of how one works with a Django development server.

pastukhov, 2015-04-01:
I mean kind of minicom/screen attached to uart or just dump  the  messages thac comming from serial port to console
For monitor and auto upload is posible to run inotify api(at least in linux), i have some examples i made for git autocommit and will post it tomorow

pastukhov, 2015-04-01:
I use mcedit :), but with inotofy is no matter what kind of editor to use.

pastukhov, 2015-04-02:
here is small script i made to auto commit with git
```bash
#!/bin/bash
#
# gitwait - watch file and git commit all changes as they happen
#
gitpath=/home/artem/build/EasyVR/python/git

while true; do

  inotifywait -qq -e CLOSE_WRITE $gitpath

  cd $gitpath; git commit -a -m "autoupdate `date +%F-%T`"

done
```
You can adopt it to nodemcu-uploader

kmpm, 2015-04-07:
@pastukhov, what would the benefit of dumping stuff to console using uploader be that screen doesn't do today?

lrozema, 2015-04-07:
Some things that I develop for nodemcu have me constantly cycle from uploading, restarting, executing and then checking the console for the result. Currently this consists of various tasks switching. Whereby when starting screen (or miniterm in my case) the nodemcu restarts because of DTR signalling. On some device that I have used the reset signals were wired differently, but with my real nodemcu it works that way. So after having a terminal connection I have to manually run the newly uploaded files. So for my workflow it would have great benefit to have everything done my nodemcu so that I can just leave it run while developing with my favorite editor.

kmpm, 2015-04-07:
So, adding that ability to uploader is just to avoid DTR signaling caused by screen? Is that a correct interpretation?

lrozema, 2015-04-07:
Well screen can't run at the same time as the uploader. So one has to switch back and forth. So for me the best implementation is to have the uploader start miniterm when done. I already have this implemented in my prototype.

kmpm, 2015-04-07:
I would not even add the functionality to start an external application when upload is done but use a script that does what I want in sequence instead and use existing software. 
Like a bash script that does inotify -> kill minicom -> uploader -> minicom instead.
So unless someone can give me a good use case for this I will close the issue.

lrozema, 2015-04-07:
Take for instance the DNS code that I have hacked on. I had to do the following:
* upload dns.lua using nodemcu uploader
* start miniterm
* enter dofile("dns.lua")
* do some resolve and see the thing crash in miniterm
* modify in dns.lua and redo the above.

In case of the inotify construction I would still have to enter the dofile manually or add this stuff as a macro in the terminal. So I guess you can close the issue.

kmpm, 2015-04-07:
ok... entering dofile manually would be a pain and adding an option to call dofile in uploader would not help that much since it might "do it's thing" before miniterm/screen/whatever had attached.
I'll wait for @pastukhov to have a chance of giving a valid use case but otherwise this will be closed.

pastukhov, 2015-04-09:
Hi all!
I'm working similar to @lrozema but. use ESPlorer instead miniterm.

So, i see nodemcu-uploader improvements like this:
```bash
nodemcu-uploader.py upload file.lua --compile --do --log
```
That will upload the script, compile it, run compiled and attach to UART to see what script print while running.

closed 2015-05-27 by kmpm