kmpm/nodemcu-uploader #14: Add uploading through telnet option?

raz123, 2015-05-11:
Most uploading utilities, including this one, seem to exclusively rely on the COM port to push files to the ESP8266. Would it be possible to add an option to nodemcu-uploader allowing to upload to an ESP8266 that is running a telnet server?

kmpm, 2015-06-04:
Technically it would be possible and I might consider it if there would be a telnet server that got a wide enough spread and had the necessary features.
Adding a telnet client to nodemcu-uploader would expand it's size somewhat but not having a standard server to talk to is my major concern.

raz123, 2015-06-05:
No worries. I have (through a fork) written a py-based telnet interface for NodeMCU. It can do binary uploads with file verification. I will be making it public soon enough.

trisk, 2015-06-07:
I just added a remote/telnet option to luatool: https://github.com/4refr0nt/luatool/pull/25
I wasn't aware of this tool previously, though it should be easy to make a similar change here. Just have to wrap the read/write calls to handle both cases (or you can convert the socket to a file stream but it will probably require additional flushing).

kmpm, 2016-01-03:
Unless a default telnet server is provided with nodemcu this will not be implemented.
Please open again if so.

closed 2016-01-03 by kmpm