kmpm/nodemcu-uploader #16: Add ability to compile only some files in an upload

egelmex, 2015-05-17:
During an upload I would like the option to only some of the uploaded files.

kmpm, 2015-06-04:
Can you think of a way of how those parameters should look on the command line. It's doable but I can't imagine a way that would not mess things up.

egelmex, 2015-06-04:
Good point, I will have a think about it.

benrussell, 2015-07-09:
./nodemcu-uploader.py upload init.lua:new_init.lc

benrussell, 2015-07-09:
or, new tool:
./nodemcu-compiler.py init.lua:init.lc

This would be my preference.

kmpm, 2015-11-15:
Ok. 
I will not make a new tool.
Detecting if a file called init.lua should be saved as init.lc could be done.
If you come with a pull request I will use it but since I am somewhat short on time I will not do it now.