kmpm/nodemcu-uploader #17: Downloads get trunkated to 1026 Bytes

jonasbjurel, 2015-06-01:
Hi I have a problem when I'm downloading files (binary or text), the files always get trunkated to 1026 Bytes.
Regards/Jonas

jonasbjurel, 2015-06-01:
As a complete newbe on Python - I didn't find the way to incorporate a python variable into a string.
But as a hint - this stupid patch made it working for 4x1024 Bytes downloads with bit exact results, this is obviously not a valid patch - but may give you a hint, and made me progress. I will read more on python the comming days and may commit a serious patch:

    def download_file(self, filename):
        bytes_read = 0
        data=""
        while True:
            if bytes_read == 0:
                d = self.exchange("file.open('" + filename + r"') print(file.seek('end', 0)) file.seek('set', 0) uart.write(0, file.read(1024)) file.close()"

            if bytes_read == 1024:
                d = self.exchange("file.open('" + filename + r"') print(file.seek('end', 0)) file.seek('set', 1024) uart.write(0, file.read(1024)) file.close()")

            if bytes_read == 2048:
                d = self.exchange("file.open('" + filename + r"') print(file.seek('end', 0)) file.seek('set', 2048) uart.write(0, file.read(1024)) file.close()")

            if bytes_read == 3072:
                d = self.exchange("file.open('" + filename + r"') print(file.seek('end', 0)) file.seek('set', 3072) uart.write(0, file.read(1024)) file.close()")

            if bytes_read == 4096:
                d = self.exchange("file.open('" + filename + r"') print(file.seek('end', 0)) file.seek('set', 4096) uart.write(0, file.read(1024)) file.close()"

            cmd, size, data_tmp = d.split('\n', 2)
            data = data + data_tmp[0:1024]
            bytes_read = bytes_read + 1024
            if bytes_read > int(size):
                brek
        data = data[0:int(size)]
        return data

kmpm, 2015-06-02:
I think this issue is actually on the lua/esp8266 end. It probably doesn't have RAM enough to read the whole file at once and then write to uart. 
This is exactly why i did a chunked upload and what you have done is one way of doing it on the download.

If you could come up with a better patch (preferably with the same protocol as upload but not necessary) I'd be more than happy to incorporate it.
Just keep in mind that you use a small enough chunk, RAM is scarce in the esp.

jonasbjurel, 2015-06-02:
So the quick resolution to this is:
============================
    def download_file(self, filename):
        chunk_size=256
        bytes_read = 0
        data=""
        while True:
            d = self.exchange("file.open('" + filename + r"') print(file.seek('end', 0)) file.seek('set', %d) uart.write(0, file.read(%d))file.close()" % (bytes_read, chunk_size))
            cmd, size, tmp_data = d.split('\n', 2)
            data=data+tmp_data[0:chunk_size]
            bytes_read=bytes_read+chunk_size
            if bytes_read > int(size):
                break
        data = data[0:int(size)]
        return data

============================
I have tried to upload this patch but failed:-(
Not really into git-hub but more into larger gerrit communities.
I thought I would need to do the following:
- Create a git branch for my change-set
- Commit my change signed
- Upload the change "git push origin download_trunc_issue"
- And then issue a "pull request"
But I fail uploading my change-set:
>> git push origin download_trunc_issue
>"ERROR: Permission to kmpm/nodemcu-uploader.git denied to jonasbjurel.
>fatal: The remote end hung up unexpectedly"

I have my ssh pub RSA key registered with my account, and Im fine contributing to other github project, including mines.
Any thoughts?
BR/Jonas

kmpm, 2015-06-03:
in github you do a fork first, of mine.
Then clone that fork or since you already did a lot of work, add that as a
remote in your clone of mine but you must name it something else than
origin.

then
git push thenewremote whateverbranch

then go to your fork on github and create the pull request.
https://help.github.com/articles/using-pull-requests/

lrozema, 2015-06-03:
Reading this and issue #18 would it make sense to implement part of the download process native into the C firmware? It is kind of trivial and I often have to first disable the active code before I have enough RAM to upload the upload code, etc.. It doesn't feel right, right? Or is that just me?

jonasbjurel, 2015-06-03:
Well this code is not residing in ESP8266, it actuallu uses the normal uart and file calls already implemented in firmware, so I dont think it would help much.

commit 25a0b8c4 (2015-06-04): Merge pull request #18 from jonasbjurel/master

Issue with download file size truncation fixed. Closes #17
closed 2015-06-04 by kmpm