kmpm/nodemcu-uploader #19: redo file download

kmpm, 2015-06-04:
I would like to see a download protocol more similar to the upload one.
And one that does not need to send the same command over and over again until all chunks are sent.

lrozema, 2015-06-04:
You mean you want to see an upload protocol similar to the download protocol? Because the download protocol is now simply a print, right?

kmpm, 2015-06-04:
Exactly.... do a loop in lua code that sends chunks of data together with size.
I made a commit (c0bbb7d3446a230011024127226c01e176f7de65) containing, what I think, is the necessary lua code.
It's not tested yet, still fiddling with the python part.

kmpm, 2016-01-25:
Issue #33 is one of the thinks that can happen without a proper download.

commit 8a61d17b (2016-01-28): Chunked download of files.

--verify=text changed to --verify=raw

Closes #19 and  #33
closed 2016-01-28 by kmpm