kmpm/nodemcu-uploader #24: Absolute paths used as file names in upload

karrots, 2015-09-06:
When passing a file to upload as an absolute path instead of just a filename the filename on the ESP module is the full path on local host.

```
$ nodemcu-uploader upload --compile /path/to/lua/file/mylua.lua
Remove /path/to/lua/file/mylua.lc
file.remove("/path/to/lua/file/mylua.lc")
stdin:1: filename too long
> 
Transfering scwifi.lua as /path/to/lua/file/mylua.lua
Error waiting for esp "recv()
stdin:1: attempt to call global 'recv' (a nil value)
> "
Compile /path/to/lua/file/mylua.lua
node.compile("/path/to/lua/file/mylua.lua")
stdin:1: filename too long
> 
Remove /path/to/lua/file/mylua.lua
file.remove("/path/to/lua/file/mylua.lua")
stdin:1: filename too long
```

kmpm, 2015-11-14:
Sometimes it's useful to have the path appended to the filename.
There could be things like...
```
public/index.html
admin/index.html
```
There could perhaps be an option to strip some levels of the path from the name, much like you can do with the tar command, but I would not like to use the basename only.

isevcik, 2017-12-06:
This also affects files specified by current dir prefix: `./file.lua`