kmpm/nodemcu-uploader #26: How to upload folders

Euphrates-Media, 2015-11-03:
I have complex project, having lots of files and folders, how can I upload them once?
Thanks.

kmpm, 2015-11-14:
The filesystem in the ESP does not support folders to start with. You will end up with a filename including the folder path. That being said, feel free to create support for uploading using wildcards.
It might even work even if I doubt it (can not test at the moment).

I have a, perhaps unnecessarily, complex set of files that I upload.
I made a Makefile, you could use whatever scripting you like that basically does the following.

```
/src
     |- init.lua
     |- httpd.lua
     |- serial.lua
/files
     |- index.html
```
For this I have a Makefile that looks like
```
.PHONY: init code files
all: init code files
init:
	#go into src because init.lua must be named just that
	cd src && nodemcu-uploader upload init.lua
code:
	nodemcu-uploader upload --compile src/httpd.lua src/serial.lua
files: 
	nodemcu-uploader upload files/index.htm
```

This will give a long list of files in root of the esp (because there arn't any folders) called.
```
init.lua
src/httpd.lc
src/serial.lc
files/index.html
```
So in my scripts I have to make sure I include the folder part into the filename as well.
If you have "many" files this might be somewhat boring to maintain the Makefile but 
you could use macros and stuff to make most of the work for you.

kmpm, 2015-11-15:
Should be tested if wildcard upload works. If not it should be implemented.

karrots, 2015-11-15:
Wildcards do work for uploads I have been using *.lua for a while now.

There is an issue with using the path+filename as the destination file name. If you want to use the upload script with absolute paths the uploaded files end up with really long names unless the uploaded file name is specified as well.

kmpm, 2015-12-28:
This is not a bug, just a matter of how you use the tool

closed 2015-12-28 by kmpm