kmpm/nodemcu-uploader #29: ImportError: cannot import name console

s8mlu, 2015-12-30:
I'm relatively new to python so perhaps I'm doing something wrong, but what version of python and/or pyserial should I be using?

samlu@d600:~/nodemcu-uploader$ uname -a
Linux d600 3.2.0-4-686-pae #1 SMP Debian 3.2.68-1+deb7u3 i686 GNU/Linux
samlu@d600:~/nodemcu-uploader$ python --version
Python 2.7.3
samlu@d600:~/nodemcu-uploader$ sudo pip install pyserial --upgrade
Requirement already up-to-date: pyserial in /usr/local/lib/python2.7/dist-packages
Cleaning up...
samlu@d600:~/nodemcu-uploader$ ./nodemcu-uploader.py 
Traceback (most recent call last):
  File "./nodemcu-uploader.py", line 43, in <module>
    from serial.tools.miniterm import Miniterm, console, NEWLINE_CONVERISON_MAP
ImportError: cannot import name console

I ended up commenting out everything after Miniterm on line 43, and set self.newline to "\r\n" on line 51 to get it to work. 

BTW this is a great tool, thanks for starting the project!

kmpm, 2016-01-03:
I was skeptical to this to start with,  there are many perfectly fine terminal applications one could use instead of trying to create another one in this utility. 

I will check if the import fails and continue without minicom support, as well as marking it as deprecated. 

P. S
You are welcome

closed 2016-01-03 by kmpm

s8mlu, 2016-01-04:
Thanks for the fix! I rebased and it's working fine without miniterm.

commit aa1e95dc (2016-01-03): Detect bad import of console and miniterm

* Closes #29
* Deprecates terminal functions