kmpm/nodemcu-uploader #31: hangs on windows

drinkypoo, 2016-01-08:
install python 2.7 for windows
use pip to install nodemcu-uploader
try to run nodemcu-uploader and it hangs:

C:\Users\drink>nodemcu-uploader.py --verbose -p COM7 -b 9600 file list
write: ;

Can't be broken, have to close cmd window
Try to connect with ESPlorer... works right away.

Doesn't work on windows python.

kmpm, 2016-01-08:
Which version of Windows and is it 32 or 64 bit python?

drinkypoo, 2016-01-08:
Win7 x64, 64 bit python. Sorry for lack of detail.

kmpm, 2016-01-08:
Nah, sorry for the lack of testing. Windows isn't my platform of choice when it comes to nodemcu development and this is actually my first time packaging a python application so... 

But improvements are comming

kmpm, 2016-01-12:
Could you please checkout the "next" branch and do a 

    python setup.py install

And then try

    nodemcu-uploader --verbose -p COM7 file list

I don't have a nodemcu to test on with me.

kmpm, 2016-01-25:
next is now merged into master

drinkypoo, 2016-01-25:
Sorry, downloaded it and tried it but now I can't get anything to talk to
my ESP-01s. I did use the windows uploader since my last test, though, and
THAT worked. I am now waiting for an ESP-01 programmer to show up and a new
serial interface to go with it so that I can rule out my hardware.

kmpm, 2016-01-25:
I have ordered some new hardware as well (it's in the mail) since every ESP module I got is now installed somewhere.

kmpm, 2016-01-28:
I can confirm it as well. 
It doesn't hang with pyserial >= 3.0 but there are other windows issues that are probably related. See #38.

What kind of serial port hardware do you have. Is it an usb adapter, on motherboard or what?

drinkypoo, 2016-01-28:
Official FTDI adapter on USB. About to try a CH340 or whatever they are
called, when it arrives.

commit 9d5de832 (2016-01-28): Do not change timeout on windows.

According to [1] so can chaning of timeout corrupt data.

1. http://sourceforge.net/p/pyserial/bugs/178/

Closes #31, Closes #38
closed 2016-01-28 by kmpm