kmpm/nodemcu-uploader #32: pySerial 3.0 API change

rahulpowar, 2016-01-22:
Ran into an issue with a fresh install.

When running with `--baud 115200`, I got an exception `AttributeError: 'Serial' object has no attribute 'setBaudrate'`

Quick search led to this link.

http://stackoverflow.com/questions/34602168/dronekit-failed-to-connect-to-dev-tty-usbmodem1411-serial-object-has-no-at

Downgrading pySerial did work but likely a fix will be needed soon.

kmpm, 2016-01-25:
* How did you install, using pip or running from a git clone?
* Which platform (win, linux, mac ...)?
* Downgrade to what, pySerial 2.7 I guess?

rahulpowar, 2016-01-25:
1. Git clone
2. OS-X
3. Yes, 2.7 as noted in the link posted

kmpm, 2016-01-25:
The solution in https://github.com/mavlink/mavlink/issues/498 seemed as an reasonable workaround.

Please pull the latest master (mainly reorganization) and try again with pySerial 3.0

kmpm, 2016-01-27:
Havent tested terminal but everything else works in the "next" branch at least

kmpm, 2016-01-28:
Fixed in df716cc874fc2d56f1867c50a93e714e9b5d7aec

closed 2016-01-28 by kmpm