kmpm/nodemcu-uploader #33: Upload verification breaks if file has "> " in it

ABonner, 2016-01-25:
Been using the uploader - nice utility!

Ran into an issue with running uploads, in that the verify step will break if there is an embedded "> " in it. `expect()` is waiting for "> " (I'm assuming the command prompt, but if it exists within the file itself it stops at that point.

Testcase file attached. I can submit a pull request once I resolve (unless you were already working on a fix to this, didn't see anything in next branch)
[testuploadfail.txt](https://github.com/kmpm/nodemcu-uploader/files/102644/testuploadfail.txt)

kmpm, 2016-01-25:
1. Thanks, and I hope it will get even better.
2. If possible (and crypto is one of the modules you have available in nodemcu) please, instead
  of just --verify run with __--verify=sha1__ as this doesn't download anything but just compares
  checksums.
3. No, I don't have anything for that in any branch and the proper way of resolving this, using the default verify method, would be  to do a download in the same chunked manner as upload to ensure that there would be support for all kinds of file content. The proper way of doing verify would actually be to use --verify=sha1 as that is a "prettier" solution and the only one to verify a file without risking it being tampered with when being transfered. sha1 calculates the checksum for the file ON the nodemcu, not just downloading the file and checking if they seem the same.

In short, if you come up with a solution that is better than --verify=sha1 (which only drawback is that it requires the crypto module) then please create a pull request on any branch you like (it will be merged into next if made against master if I haven't merged the two branches by then). Any help is appreciated.

kmpm, 2016-01-25:
Next is now merged into master

kmpm, 2016-01-28:
Ooops, sha1 verify as file-size limit. See #40

kmpm, 2016-01-28:
use --verify=raw and it should work

closed 2016-01-28 by kmpm

ABonner, 2016-01-29:
Thanks, been a busy week at work so I will take a look this weekend to confirm.