kmpm/nodemcu-uploader #46: --verify=raw results in error

romanchyla, 2016-02-22:
The error is essentially inside the read_chunk function, which doesn't expect the buffer to be zero - after 100us, the port did not return any data - so it fails. However, it will succeed only if I increase the timeout by 1s! It is possible that this device is particularly slow (?) - however, that hardcode limit of 0.0001 seems be too tight

[btw: I got into these troubles because I noticed that very often one of my files gets only partially uploaded; this has caused many bugs and started to use --verify raw]

If I increase the timeout, the data gets read - this is on nodemcu Lolin v3
```
> =node.info()
1	4	0	1013222	1458400	4096	2	40000000
> =node.chipid()
1013222
> =node.flashid()
1458400
> 
```

stacktrace:
```
/dvt/workspace2/company/hw/esp8266/lua /dvt/workspace2/company/factory
opening port /dev/ttyUSB0 with 9600 baud
Changing communication to 19200 baud
Preparing esp for transfer.
Preparation already done. Not adding functions again.
Transfering iotzfutils.lua as iotzfutils.lua
Verifying...
receiveing iotzfutils.lua
Traceback (most recent call last):
  File "/dvt/workspace2/company/python/bin/nodemcu-uploader", line 9, in <module>
    load_entry_point('nodemcu-uploader==0.2.2', 'console_scripts', 'nodemcu-uploader')()
  File "/dvt/workspace2/company/python/local/lib/python2.7/site-packages/nodemcu_uploader/main.py", line 245, in main_func
    args.restart)
  File "/dvt/workspace2/company/python/local/lib/python2.7/site-packages/nodemcu_uploader/main.py", line 48, in operation_upload
    uploader.write_file(f, d, verify)
  File "/dvt/workspace2/company/python/local/lib/python2.7/site-packages/nodemcu_uploader/uploader.py", line 271, in write_file
    data = self.download_file(destination)
  File "/dvt/workspace2/company/python/local/lib/python2.7/site-packages/nodemcu_uploader/uploader.py", line 217, in download_file
    chunk, buffer = self.read_chunk(buffer)
  File "/dvt/workspace2/company/python/local/lib/python2.7/site-packages/nodemcu_uploader/uploader.py", line 355, in read_chunk
    if buffer[0] != BLOCK_START or len(buffer) < 130:
IndexError: string index out of range
```

kmpm, 2016-02-22:
There are some more timeout adjustments comming in #43 that will solve some of this.
But I will have another look as well.

kmpm, 2016-03-23:
could you please test this again, using the "next" branch

kmpm, 2016-03-23:
or master since next (v0.3.0) is merged into master

kmpm, 2016-08-11:
If I don't get any more feedback on this I will assume that the issue was fixed in a later version and will close the Issue.

closed 2017-06-13 by kmpm