kmpm/nodemcu-uploader #47: uploading a file results in a PANIC

l29ah, 2016-02-23:
```
‰ nodemcu-uploader --verbose --port /dev/ttyUSB1 -b 9600 upload init.lua
opening port /dev/ttyUSB1 with 9600 baud
getting in sync with LUA
write: ;

expect returned: `;
stdin:1: unexpected symbol near ';'
> `
write: print("%sync%");

expect returned: `print("%sync%");
%sync%
> `
Preparing esp for transfer.
write: print(recv_block)

expect returned: `print(recv_block)
function: 3fff0c30
> `
write: print(recv_name)

expect returned: `print(recv_name)
function: 3fff0c58
> `
write: print(recv)

expect returned: `print(recv)
function: 3fff1600
> `
write: print(shafile)

expect returned: `print(shafile)
function: 3fff17b0
> `
write: print(send_block)

expect returned: `print(send_block)
function: 3fff17d8
> `
write: print(send_file)

expect returned: `print(send_file)
function: 3fff0c80
> `
write: print(send)

expect returned: `print(send)
nil
> `
write: function recv_block(d)

expect returned: `function recv_block(d)
>> `
write: if string.byte(d,1) == 1 then

expect returned: `if string.byte(d,1) == 1 then
>> `
write: size=string.byte(d,2)

expect returned: `size=string.byte(d,2)
>> `
write: uart.write(0,'\006')

expect returned: `uart.write(0,'\006')
>> `
write: if size > 0 then

expect returned: `if size > `
write: file.write(string.sub(d,3,3+size-1))

expect returned: `0 then
>> `
write: else

expect returned: `file.write(string.sub(d,3,3+size-1))
>> `
write: file.close()

expect returned: `else
>> `
write: uart.on('data')

expect returned: `file.close()
>> `
write: end

expect returned: `uart.on('data')
>> `
write: else

expect returned: `end
>> `
write: uart.write(0,'\021' .. d)

expect returned: `else
>> `
write: uart.on('data')

expect returned: `uart.write(0,'\021' .. d)
>> `
write: end

expect returned: `uart.on('data')
>> `
write: end

expect returned: `end
>> `
write: function recv_name(d) d=string.gsub(d,'\000','') file.remove(d) file.open(d,'w') uart.on('data',130,recv_block,0) uart.write(0,'\006') end

expect returned: `end
> `
write: function recv() uart.on('data','\000',recv_name,0) uart.write(0,'C') end

expect returned: `function recv_name(d) d=string.gsub(d,'\000','') file.remove(d) file.open(d,'w') uart.on('data',130,recv_block,0) uart.write(0,'\006') end
> `
write: function shafile(f) file.open(f,"r") print(crypto.toHex(crypto.hash("sha1",file.read()))) file.close() end

expect returned: `function recv() uart.on('data','\000',recv_name,0) uart.write(0,'C') end
> `
write: function send_block(d) l=string.len(d) uart.write(0,'\001' .. string.char(l) .. d .. string.rep('#',128 - l)) return l end

expect returned: `function shafile(f) file.open(f,"r") print(crypto.toHex(crypto.hash("sha1",file.read()))) file.close() end
> `
write: function send_file(f) file.open(f) s=file.seek('end',0) p=0 uart.on('data',1,function(data)

expect returned: `function send_block(d) l=string.len(d) uart.write(0,'\001' .. string.char(l) .. d .. string.rep('#',128 - l)) return l end
> `
write: if data == '\006' and p<s then file.seek('set',p) p=p+send_block(file.read(128)) else

expect returned: `function send_file(f) file.open(f) s=file.seek('end',0) p=0 uart.on('data',1,function(data)
>> `
write: send_block('') file.close() uart.on('data') print('interrupted') end end,0) uart.write(0,f .. '\000')

expect returned: `if data == '\006' and p<s then file.seek('set',p) p=p+send_block(file.read(128)) else
>> `
write: end

expect returned: `send_block('') file.close() uart.on('data') print('interrupted') end end,0) uart.write(0,f .. '\000')
>> `
write: function send(f) uart.on('data',1,function (data)

expect returned: `end
> `
write: uart.on('data') if data == 'C' then send_file(f) else print('transfer interrupted') end end,0)

expect returned: `function send(f) uart.on('data',1,function (data)
>> `
write: end

expect returned: `uart.on('data') if data == 'C' then send_file(f) else print('transfer interrupted') end end,0)
>> `
Transfering init.lua as init.lua
write: recv()

expect returned: `end
> recv()
C> `
sending destination filename "init.lua"
write binary: 69:6e:69:74:2e:6c:75:61:00
waiting for ack
ack read 06
sending 256 bytes in init.lua
writing 128 bytes chunk
packet size 130
write: conn=net.createConnection(net.TCP, 0)
conn:on("connection", function(sck, c)
    conn:send("hey\n")
    conn:close()
end)
functi
waiting for ack
ack read 06
writing 128 bytes chunk
packet size 130
write: on tcp_send()
    conn:connect(4242, "192.168.1.2")
end
wifi.sta.eventMonReg(wifi.STA_GOTIP, tcp_send)
wifi.sta.eventMonStart()

waiting for ack
ack read 50
expect returned: `ANIC: unprotected error in call to Lua API (stdin:6: open a file first)
c_ÇRSöâF*Söf*jîXÿ

NodeMCU custom build by frightanic.com
	branch: master
	commit: c8037568571edb5c568c2f8231e4f8ce0683b883
	SSL: false
	modules: file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2016-02-22 22:14
 powered by Lua 5.1.4 on SDK 1.4.0
> `
Bad chunk response "ANIC: unprotected error in call to Lua API (stdin:6: open a file first)
c_ÇRSöâF*Söf*jîXÿ

NodeMCU custom build by frightanic.com
	branch: master
	commit: c8037568571edb5c568c2f8231e4f8ce0683b883
	SSL: false
	modules: file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2016-02-22 22:14
 powered by Lua 5.1.4 on SDK 1.4.0
> " 41:4e:49:43:3a:20:75:6e:70:72:6f:74:65:63:74:65:64:20:65:72:72:6f:72:20:69:6e:20:63:61:6c:6c:20:74:6f:20:4c:75:61:20:41:50:49:20:28:73:74:64:69:6e:3a:36:3a:20:6f:70:65:6e:20:61:20:66:69:6c:65:20:66:69:72:73:74:29:0d:0a:63:5f:c7:52:53:f6:e2:46:2a:53:f6:66:2a:03:6a:ee:58:ff:0d:0a:0d:0a:4e:6f:64:65:4d:43:55:20:63:75:73:74:6f:6d:20:62:75:69:6c:64:20:62:79:20:66:72:69:67:68:74:61:6e:69:63:2e:63:6f:6d:0d:0a:09:62:72:61:6e:63:68:3a:20:6d:61:73:74:65:72:0d:0a:09:63:6f:6d:6d:69:74:3a:20:63:38:30:33:37:35:36:38:35:37:31:65:64:62:35:63:35:36:38:63:32:66:38:32:33:31:65:34:66:38:63:65:30:36:38:33:62:38:38:33:0d:0a:09:53:53:4c:3a:20:66:61:6c:73:65:0d:0a:09:6d:6f:64:75:6c:65:73:3a:20:66:69:6c:65:2c:67:70:69:6f:2c:6e:65:74:2c:6e:6f:64:65:2c:74:6d:72:2c:75:61:72:74:2c:77:69:66:69:0d:0a:20:62:75:69:6c:64:20:09:62:75:69:6c:74:20:6f:6e:3a:20:32:30:31:36:2d:30:32:2d:32:32:20:32:32:3a:31:34:0d:0a:20:70:6f:77:65:72:65:64:20:62:79:20:4c:75:61:20:35:2e:31:2e:34:20:6f:6e:20:53:44:4b:20:31:2e:34:2e:30:0d:0a:3e:20
All done!
closing port
```

kmpm, 2016-02-24:
Some questions...

* What OS do you have?
* nodemcu-uploader --version gives you what version?
* since you flashed the device with the new firmware, have you formated the file system? 
  
  __nodemcu-upload file format__

kmpm, 2016-03-25:
please test with a version of nodemcu-uploader >=0.3.1
Issue #48 might have had something to do with it.

kmpm, 2016-08-11:
If I don't get any more feedback on this I will assume the issue is fixed with a later version of the tool and I will close the issue.

st3fan, 2016-08-15:
In my case the same thing happened. (I am new to all this) - The `nodemcu-uploader file format` did the trick and got me past this same issue.

kmpm, 2016-08-15:
@st3fan, that is the recommended procedure with nodemcu firmware. To always format the file system after a reflash. 

I should perhaps make a note of that in my Readme as well.

marcelstoer, 2017-03-04:
> the recommended procedure with nodemcu firmware

That's your personal recommendation, right? The firmware project doesn't give any such recommendations. http://nodemcu.readthedocs.io/en/latest/en/flash/#upgrading-firmware even implies that the firmware will format the file system automatically when needed.

kmpm, 2017-06-13:
It's my personal recommendation but the documentation says that 

> The NodeMCU flash file system __MAY__ need to be reformatted, particularly if its address has changed because the new firmware is different in size from the old firmware.

(my emphasis on MAY)
I have had so many issues with this not being done automatically so that it's one of the first things I do if something isn't working.

closed 2017-06-13 by kmpm