kmpm/nodemcu-uploader #48: PANIC: bad argument #1 to 'open' (filename invalid)

Christoph-D, 2016-03-09:
I get a "filename invalid" with the current dev branch of nodemcu-firmware (nodemcu/nodemcu-firmware@f5acf996208515eef3ef22c6ed9846953640e8a6) and the current master branch of nodemcu-uploader.
````
Transfering alarms.lua as alarms.lua
write: recv()

expect returned: `file.remove("alarms.lc")
> recv()
C> `
sending destination filename "alarms.lua"
write binary: 61:6c:61:72:6d:73:2e:6c:75:61:00
waiting for ack
ack read 50
did not ack destination filename
Compile alarms.lua
write: node.compile("alarms.lua")

expect returned: `ANIC: unprotected error in call to Lua API (stdin:1: bad argument #1 to 'open' (filename invalid))
````
I formatted the file system after flashing the firmware.

7c7026491a4a730965903d7eb848a47f4e628195 fixes it for me, but I don't know why (no time to do a git bisect on nodemcu-firmware now).  That is why I didn't open a pull request.  Nonetheless, I hope the patch points in the right direction.

kmpm, 2016-03-23:
I will not change the zero byte marker... It must be firmware related and for the moment I do not have much time to support the dev branch of the firmware. Help is much appreciated.

visvirial, 2016-03-24:
Same here.

It seems bug in string.gsub() API in NodeMCU firmware...

I tested it with this code:
```
a = '\000'
print(a == '\000', 'should be "true"')
print(a == '',     'should be "false"')

b = string.gsub(a, '\000', '')
print(b == '\000', 'should be "false"')
print(b == '',     'should be "true"')
```

When running on my local machine (Ubuntu, Lua5.3), I got a correct result.
```
$ lua5.3 test.lua 
true	should be "true"
false	should be "false"
false	should be "false"
true	should be "true"
```

But if I run this code on NodeMCU (via `nodemcu-uploader terminal` command) I got
```
> a = '\000'
> print(a == '\000', 'should be "true"')
true	should be "true"
> print(a == '',     'should be "false"')
false	should be "false"
> 
> b = string.gsub(a, '\000', '')
> print(b == '\000', 'should be "false"')
true	should be "false"
> print(b == '',     'should be "true"')
false	should be "true"
```

From the test code above, it seems NodeMCU dev version does not handle NULL byte correctly in string.gsub().

kmpm, 2016-03-24:
I will have to rework the code to make it work with some changes in upstream.
gsub does not replace _all_ strings, nice.

kmpm, 2016-03-24:
Please test the latest "next" as I just commited some changes that might work.

commit b7efdb83 (2016-03-25): even better way of getting null terminated filename

now works with dev firmware closes #48
closed 2016-03-25 by kmpm