kmpm/nodemcu-uploader #51: Change default baudrate when the new speed is in the default firmware

kmpm, 2016-03-25:
In the dev branch of nodemcu-firmware the default baudrate is now 115200.

When that gets merged into master, the default START_BAUD needs to be changed in uploader

pjsg, 2016-04-24:
The default configuration of nodemcu-firmware:dev (shortly to be master) now includes the auto-baudrate detection. This should enable the uploader to work with any speed. However, it does require a few characters to enable it to lock on to the speed.

I suspect that putting the following before the line
```
                self.__exchange(';') # Get a defined state
```

in uploader.py to something like
```
                self.__exchange('UUUUUUUU') # Send enough characters for auto-baud
                time.sleep(0.15) # Wait for autobaud timer to expire
```

kmpm, 2016-04-24:
Good one
Den 24 apr. 2016 7:47 em skrev "Philip Gladstone" <notifications@github.com

kmpm, 2016-07-29:
according to  https://github.com/nodemcu/nodemcu-firmware/docs/en/modules/uart.md 
> The default rate is 115,200 bps. In addition, auto-baudrate detection is enabled for the
> first two minutes after platform boot. This will cause a switch to the correct baud rate
> once a few characters are received.

So if more that 2 minutes have passed the auto-baudrate does not work any more, so falling back to a sensible default would be good either way.

commit c834cc31 (2016-07-29): Change default baudrate

* Closes #51
* implements auto-baudrate detection as mentioned by @matgoebl
* updates usage
closed 2016-07-29 by kmpm