kmpm/nodemcu-uploader #52: Syntax error after run

sturchaninov, 2016-03-25:
When running without arguments from Windows with Python 3.5.1 I see syntax error message:
...ges\nodemcu_uploader\uploader.py", line 348
    print 'buffer size:', len(buffer)
                       ^
SyntaxError: invalid syntax

Seems it happens because of old print function syntax used (print "statement" instead of print("...")) in:
https://github.com/kmpm/nodemcu-uploader/blob/master/nodemcu_uploader/uploader.py#L387
Please fix if I'm right.

kmpm, 2016-03-25:
Uploader is currently not supported under python 3 and there are
unfortunately no work in progress to support it.

It will happen, but I don't know when.

kmpm, 2017-06-13:
This should be working since version 0.4.3

closed 2017-06-13 by kmpm