kmpm/nodemcu-uploader #53: nodemcu-uploader not working with current NodeMCU DEV build

skycoders, 2016-03-26:
Hi, I have just flashed the latest DEV build on my ESP (custom built). Even though I adjusted the default baud rate back to 9600 in the firmware, I cannot use the "old" nodemcu-uploader". I do not get an ACK for the filename, and the console later shows "bad argument #1 to 'open'...)

I would have tried with the current version of nodemcu-uploader, but that even refuses to connect to the usb-tty device, even though I can flash firmware through it without problems.

kmpm, 2016-03-26:
I used it with dev from 2 days ago but I'll check again.

kmpm, 2016-03-26:
With uploader v0.3.1 I did the following on a adafruit Huzzah with and FTDI serial converter on my USB bus. I just created a custom build using nodemcu-build.com from the dev branch and added the crypto module (to enable sha1 verification)

```
$ nodemcu-uploader -B 115200 upload README.md --verify=sha1
opening port /dev/ttyUSB0 with 115200 baud
Preparing esp for transfer.
Transferring README.md as README.md
Verifying using sha1...
Remote SHA1: 7d5edeb47f2a1fd9130c6e472af642a69d36013d
Local SHA1: 7d5edeb47f2a1fd9130c6e472af642a69d36013d
Verification successful. Checksums match
All done!
```

The -B or --start_baud lets you change the "default" baudrate that uploader uses. This should match the default baudrate of the firmware.

There must be something else going on with your device or using an old version of nodemcu-uploader. There is a known issue, #48, with older versions of nodemcu-uploader that looks like your problem but that is fixed in the latest. 
I'm closing this issue but feel free to reopen if it really is something wrong with v0.3.1 and dev firmware. If it's something else, please create a new issue.

closed 2016-03-26 by kmpm