kmpm/nodemcu-uploader #54: sync fail on baud change

2114L3, 2016-04-05:
thanks for the great tool 

hit a bug with the baud change. 

using;
====================
linuxmint 17.3 mate, 
python 2.7.6, 
pyserial 3.0.1
pip install nodemcu-uploader (version 0.3.1)
ESP-01 module running nodemcu 0.9.5 build 20150318

running;
====================
nodemcu-uploader --verbose --baud 19200 upload hello.lua

returns;
====================
```
opening port /dev/ttyUSB0 with 9600 baud
getting in sync with LUA
write: ;

expect returned: `;
stdin:1: unexpected symbol near ';'
> `
write: print("%sync%");

expect returned: `print("%sync%");
%sync%
> `
Changing communication to 19200 baud
write: uart.setup(0,19200,8,0,1,1)

getting in sync with LUA
write: ;

expect returned: `> `
write: print("%sync%");

expect returned: `�print("%sync%");
stdin:1: unexpected symbol near '�'
> `
Traceback (most recent call last):
  File "/usr/local/bin/nodemcu-uploader", line 9, in <module>
    load_entry_point('nodemcu-uploader==0.3.1', 'console_scripts', 'nodemcu-uploader')()
  File "/usr/local/lib/python2.7/dist-packages/nodemcu_uploader/main.py", line 243, in main_func
    uploader = Uploader(args.port, args.baud, start_baud=args.start_baud)
  File "/usr/local/lib/python2.7/dist-packages/nodemcu_uploader/uploader.py", line 75, in __init__
    __sync()
  File "/usr/local/lib/python2.7/dist-packages/nodemcu_uploader/uploader.py", line 67, in __sync
    raise DeviceNotFoundException('Device not found or wrong port')
nodemcu_uploader.exceptions.DeviceNotFoundException: Device not found or wrong port
```

hack;
==================
insert at line 62 an second
`self.__exchange(';')`
and the issue goes away as the ; command clears the unexpected symbol and allows the print statement to run without error. clearing buffers after the ; exchange and before the print sync doesnt have any effect.

note;
====================
no issue if start_baud and baud match as the sync routine it bypassed
other speeds have the same issue. used 19200 to decrease likely hood of high speed issues as a troubleshooting step.

thanks!

kmpm, 2016-04-05:
Interesting,

Thanks for the find. I will do some work this weekend and try to figure out
why...

2114L3, 2016-04-06:
i was thinking it might be the FTDI serial TTL that i was using. 
but i swapped to a CP2104 using the default cp210x driver and i have the same issue. 

**FTDI details;**
```
Bus 003 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0403 Future Technology Devices International, Ltd
  idProduct          0x6001 FT232 Serial (UART) IC
  bcdDevice            6.00
  iManufacturer           1 FTDI
  iProduct                2 FT232R USB UART
  iSerial                 3 A9Q5PVFJ
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               90mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 FT232R USB UART
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)

T:  Bus=03 Lev=01 Prnt=01 Port=01 Cnt=02 Dev#=  4 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=0403 ProdID=6001 Rev=06.00
S:  Manufacturer=FTDI
S:  Product=FT232R USB UART
S:  SerialNumber=A9Q5PVFJ
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=90mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=ff Driver=ftdi_sio

```

**cp2104 details;**
```
Bus 003 Device 008: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x10c4 Cygnal Integrated Products, Inc.
  idProduct          0xea60 CP210x UART Bridge / myAVR mySmartUSB light
  bcdDevice            1.00
  iManufacturer           1 Silicon Labs
  iProduct                2 CP2104 USB to UART Bridge Controller
  iSerial                 3 0100AF84
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 CP2104 USB to UART Bridge Controller
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)

T:  Bus=03 Lev=01 Prnt=01 Port=01 Cnt=02 Dev#=  4 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=0403 ProdID=6001 Rev=06.00
S:  Manufacturer=FTDI
S:  Product=FT232R USB UART
S:  SerialNumber=A9Q5PVFJ
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=90mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=ff Driver=ftdi_sio
```

kmpm, 2016-05-17:
I haven't been able to reproduce your findings with any of the later versions of nodemcu firmware.
It's running fine with a build from http://nodemcu-build.com/ using the master branch and speeds at least up to 115200 on my FTDI based _Bus 003 Device 002: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC_ cable.

Please test again with a more resent version of the firmware and reopen the issue if it still exists.

closed 2016-05-17 by kmpm