kmpm/nodemcu-uploader #60: Device not found or wrong port

idreamerhx, 2016-06-07:
Hi,

Thanks for the nice project.

I run nodemcu-uploader on raspberry pi 2. nodemcu is connected to usb.

Bus 001 Device 009: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light

/dev/ttyUSB0 exists.

esptool runs ok.

[root@pi nodemcu-uploader]# python2 nodemcu-uploader.py  file list
opening port /dev/ttyUSB0 with 9600 baud
Traceback (most recent call last):
  File "nodemcu-uploader.py", line 10, in <module>
    main.main_func()
  File "/opt/nodemcu-uploader/nodemcu_uploader/main.py", line 250, in main_func
    uploader = Uploader(args.port, args.baud, start_baud=args.start_baud)
  File "/opt/nodemcu-uploader/nodemcu_uploader/uploader.py", line 68, in __init__
    __sync()
  File "/opt/nodemcu-uploader/nodemcu_uploader/uploader.py", line 66, in __sync
    raise DeviceNotFoundException('Device not found or wrong port')
nodemcu_uploader.exceptions.DeviceNotFoundException: Device not found or wrong port

don41382, 2016-07-23:
same here

```screen /dev/tty.wch8266 115200```

works without any problem

```nodemcu-uploader --start_baud 115200 --port /dev/tty.wch8266 upload init.lua```

ends with

```
Traceback (most recent call last):
  File "/usr/local/bin/nodemcu-uploader", line 11, in <module>
    sys.exit(main_func())
  File "/usr/local/lib/python2.7/site-packages/nodemcu_uploader/main.py", line 250, in main_func
    uploader = Uploader(args.port, args.baud, start_baud=args.start_baud)
  File "/usr/local/lib/python2.7/site-packages/nodemcu_uploader/uploader.py", line 68, in __init__
    __sync()
  File "/usr/local/lib/python2.7/site-packages/nodemcu_uploader/uploader.py", line 66, in __sync
    raise DeviceNotFoundException('Device not found or wrong port')
nodemcu_uploader.exceptions.DeviceNotFoundException: Device not found or wrong port
```

kmpm, 2016-07-29:
Please test again with v0.4.0 of nodemcu-uploader.
And you should be able to skip the start baud flag as well.

mk30, 2016-08-08:
i experienced the same error as the above commenters, so i installed via the git-clone method instead of via pip. unfortunately when i did "python ./nodemcu-uploader.py" i got

`usage: nodemcu-uploader [-h] [--verbose] [--version] [--port PORT]
                        [--baud BAUD] [--start_baud START_BAUD]
                        [--timeout TIMEOUT]
                        {upload,exec,download,file,node,terminal} ...
nodemcu-uploader: error: too few arguments
`

kmpm, 2016-08-09:
Then please try "python ./nodemcu-upload.py file list" or any other command that you want the software to do.
`python ./nodemcu-upload.py upload nodemcy-upload.py` for example would try to upload the python script to the device. Not so useful but works as a test.

mk30, 2016-08-09:
thank you. unfortunately i got the same error.

i ran:
`sudo python ~/dev/nodemcu-uploader/nodemcu-uploader.py upload esp_ap_webserver.ino`

which gave me:

> Traceback (most recent call last):
>   File "/home/marina/dev/nodemcu-uploader/nodemcu-uploader.py", line 10, in <module>
>     main.main_func()
>   File "/home/marina/dev/nodemcu-uploader/nodemcu_uploader/main.py", line 250, in main_func
>     uploader = Uploader(args.port, args.baud, start_baud=args.start_baud)
>   File "/home/marina/dev/nodemcu-uploader/nodemcu_uploader/uploader.py", line 71, in __init__
>     __sync()
>   File "/home/marina/dev/nodemcu-uploader/nodemcu_uploader/uploader.py", line 69, in __sync
>     raise DeviceNotFoundException('Device not found or wrong port')
> nodemcu_uploader.exceptions.DeviceNotFoundException: Device not found or wrong port

kmpm, 2016-08-09:
What version of the nodemcu firmware do you have on your device. What kind of serial hardware do you have for connecting the device to your pc?

And ino files, aren't they for arduino? Nodemcu will not be able to use them anyway.

mk30, 2016-08-10:
i have the default float build from here: http://nodemcu-build.com/

i connect via usb (i have an esp that has a usb port).

i am trying to upload [this code](http://arduino-er.blogspot.com/2016/05/nodemcuesp8266-act-as-ap-access-point_3.html), which the author seems to be using on their nodemcu esp. i will try a different file (not ino).

planza5, 2016-08-11:
Same Error...

My chipset NodeMCU DEVKIT v0.9 05/12/2014
firmware from http://nodemcu-build.com/
connecting usb from my laptop (windows 8.1)

I can upload Arduino sketchs running fine so chip is not bricked. When I try uploading a lua script:

```
C:\Apps\nodemcu>nodemcu-uploader --verbose --port=COM6 upload wifi.lua
opening port COM6 with 115200 baud
getting in sync with LUA
write: UUUUUUUUUUUU

write: ;

expect returned: ``
Traceback (most recent call last):
  File "C:\Python27\Scripts\nodemcu-uploader-script.py", line 9, in <module>
    load_entry_point('nodemcu-uploader==0.4.0', 'console_scripts', 'nodemcu-uplo
ader')()
  File "c:\python27\lib\site-packages\nodemcu_uploader\main.py", line 250, in ma
in_func
    uploader = Uploader(args.port, args.baud, start_baud=args.start_baud)
  File "c:\python27\lib\site-packages\nodemcu_uploader\uploader.py", line 71, in
 __init__
    __sync()
  File "c:\python27\lib\site-packages\nodemcu_uploader\uploader.py", line 69, in
 __sync
    raise DeviceNotFoundException('Device not found or wrong port')
nodemcu_uploader.exceptions.DeviceNotFoundException: Device not found or wrong p
ort
```

kmpm, 2016-08-11:
What happens if you connect to the device using a terminal (like putty, hyperterm or similar). Can you get some communication up and working then, get the prompt at least?

kmpm, 2016-08-11:
@mk30, if you do a `lsusb` could you please find out which chipset the usb to serial converter on your device might be?

planza5, 2016-08-11:
If I connect Putty to the device....Frozen...No text output from device and I can't input text. If I connect a second Putty session say 'the port is in use'

I can't run the 'lussb' command because my SO is Windows 8.1. however using USBDevView I can see this 

[https://s10.postimg.org/rwfz0ukpl/usb.jpg](url)

May be Im using a wrong firmware and this fact impacts in serial communication? My nodeMCU device is v0.9 and the actual device version is v1.0 

Yesterday I tried luatool and LuaLoader. I cant upload lua files...other traces, similar error. May be the error is not in nodemcu-uploader

kmpm, 2016-08-12:
@planza5, if you don't get a prompt then this tool can't help you since it use the same interface for its own work. 
I would try to reflash with nodemcu firmware.

mk30, 2016-08-14:
@kmpm 

i am able to connect to the device via screen with this command: screen /dev/ttyUSB0 9600. i get a bunch of unreadable characters when i do that.

this is what i get from lsusb:

```
Bus 003 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
```

kmpm, 2016-08-15:
@mk30 if you use 115200 instead of 9600 when you connect using screen, do you get anything readable then?

oyooyo, 2016-09-04:
I'm experiencing the same problem. I didn't use my ESP8266 for probably 2-3 months, so today when I wanted to use it again, I started by downloading and installing the latest versions of nodemcu-firmware (dev branch) and nodemcu-uploader.

I could access my ESP8266 just fine using ESPlorer , but when I tried to upload a file using nodemcu-uploader, I constantly got that _"Device not found or wrong part"_ error as well. It worked perfectly fine the last time I used nodemcu-uploader, so it seems that either nodemcu-firmware or nodemcu-uploader has meanwhile seen some change that caused this problem.

I tried to fix the problem by myself and started by enabled DEBUG log output.
Here's the output:
```
> python2.7 nodemcu-uploader.py file list

opening port /dev/ttyUSB0 with 115200 baud
getting in sync with LUA
write: UUUUUUUUUUUU

write: ;

expect returned: `rll[some binary "garbage", probably because of wrong baudrate]

NodeMCU 1.5.4.1 build unspecified powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
lua: cannot open init.lua
> `
write: print("%sync%");

expect returned: `;
pstdin:1: unexpected symbol near ';'
> rint("%sync%");
stdin:1: attempt to call global 'rint' (a nil value)
stack traceback:
        stdin:1: in main chunk
> `
Traceback (most recent call last):
  File "nodemcu-uploader.py", line 10, in <module>
    main.main_func()
  File "/data/project/esp8266/0.0.2/software/nodemcu-uploader/oyo-0.0.1/nodemcu_uploader/main.py", line 250, in main_func
    uploader = Uploader(args.port, args.baud, start_baud=args.start_baud)
  File "/data/project/esp8266/0.0.2/software/nodemcu-uploader/oyo-0.0.1/nodemcu_uploader/uploader.py", line 74, in __init__
    __sync()
  File "/data/project/esp8266/0.0.2/software/nodemcu-uploader/oyo-0.0.1/nodemcu_uploader/uploader.py", line 72, in __sync
    raise DeviceNotFoundException('Device not found or wrong port')
nodemcu_uploader.exceptions.DeviceNotFoundException: Device not found or wrong port
```
I played around a bit, and added a short delay in __init() in uploader.py:
```
self._port.setRTS(False)
self._port.setDTR(False)
time.sleep(0.5)
```
With that delay added, I got the following debug output instead:
```
> python2.7 nodemcu-uploader.py file list

opening port /dev/ttyUSB0 with 115200 baud
getting in sync with LUA
write: UUUUUUUUUUUU

write: ;

expect returned: `UUUUUUUUUUUU
>> `
write: print("%sync%");

expect returned: `;
stdin:2: '=' expected near ';'
> print("%sync%");
%sync%
> `
Listing files
write: for key,value in pairs(file.list()) do print(key,value) end

expect returned: `for key,value in pairs(file.list()) do print(key,value) end
init.lua        380
> `
for key,value in pairs(file.list()) do print(key,value) end
init.lua        380
> 
closing port
```

The output looks to me as if there's still some problem with the syncing and the additional time.sleep() was just a lucky but somehow working workaround, but at least nodemcu-uploader seems to work again.

harambeOut, 2016-10-01:
I've found the problem was in `/usr/local/lib/python2.7/dist-packages/nodemcu_uploader/uploader.py` line 61:

`time.sleep(0.15) # Wait for autobaud timer to expire`

0.15 wait is not enough. I've put **0.30** and it works fine.

closed 2016-10-01 by kmpm

Integ, 2017-06-21:
@harambeOut I have to make it **0.60** on my windows.

oyooyo, 2017-10-29:
I had constant problems uploading files to my ESP8266. 80% of the time uploading a file just wouldn't work. I had to repeat the process over and over until I would be lucky.

After increasing
`time.sleep(0.3) # Wait for autobaud timer to expire`
to
`time.sleep(0.5) # Wait for autobaud timer to expire`
that problem seems completely solved.

Since the current hardcoded value of 0.3 seems to be too low for other people as well (@Integ), what about increasing it, or allow to specify a value from the command line?

samdup123, 2018-06-29:
i changed mine to sleep for .8 and still experience the error

kmpm, 2018-06-29:
Should perhaps remove autobaud completely. Have been causing problems ever
since it was included.

commit ab532ac4 (2016-10-01): change autobaud timeout

* closes #60