kmpm/nodemcu-uploader #65: I wrote a nice script to go along with this

pmpbar, 2017-01-04:
```bash
#!/bin/bash

if hash nodemcu-uploader 2>/dev/null; then
  if [[ -z $1 ]]; then
    echo "Usage: [sudo] ./esp_tool.sh DEVICE"
  else
    port=$1
    clear
    while IFS="" read -r -e -d $'\n' -p '$ ' commands; do
      history -s "$commands"
      nodemcu-uploader --port $port $commands
    done
  fi
else
  echo "Please install nodemcu-uploader to use this script."
  echo "$ pip install nodemcu-uploader"
fi
```

It just is run one time and allows you to run commands multiple times. It has history so you can use ^P or the up/down arrows to scroll through your history.

If you like it, maybe put it in the repo! 😄

kmpm, 2017-06-13:
My guess is that this file should be named esp_tool.sh.
Is that correct?

pmpbar, 2017-06-13:
Yeah, it really doesn't matter haha. I forgot to take that out. Whatever you want!