kmpm/nodemcu-uploader #66: only compatible with python2?

agauniyal, 2017-01-08:
I am on a linux distro that sets python3 rather than it's old version as default. Was confused why it wasn't running properly - 
```
Traceback (most recent call last):
  File "./nodemcu-uploader.py", line 6, in <module>
    from nodemcu_uploader import main
  File "/home/agauniyal/projects/nodemcu-uploader/nodemcu_uploader/main.py", line 76
    print 'sources', sources
                  ^
SyntaxError: Missing parentheses in call to 'print'
```

should probably add some warnings saying only compatible with python2

biberesser, 2017-01-29:
I had the exact same problem.
This requirement should clearly be documented, and the script should verify its requirements and provide with a meaningful error message if that fails.
A Python 3 version of the script would ne nice, too.

saintcrawler, 2017-04-21:
Got it working in Python 3.6. At least it can upload and do raw verification. But it fails in about 8 times of 10 with different garbage in response from ESP, but I think it's hardware problems :).
If someone is interested in it, I can make a PR.

agauniyal, 2017-04-21:
@saintcrawler Yes please, most systems are defaulting to python3 atm so it would be a good change. I'll make some time to test it as well.

saintcrawler, 2017-04-22:
@agauniyal You can test it now

kmpm, 2017-06-01:
closed with bca98f6a44c970fdb0a6974acdb4732d876788c6 but it's not yet published in pypi

closed 2017-06-01 by kmpm