kmpm/nodemcu-uploader #67: Add Ability to Set Hostname in CLI When Uploading Files

liv3dn8as, 2017-04-05:
I think that it would be nice to be able to set the hostname at time of uploading. Having some sort of argument like --hostname would be nice. However, I just realized that when setting the hostname it doesn't stick after the device has been rebooted. Only if it is in the actual code. Not sure if it even makes sense then as I doubt you could really append the actual code. Maybe my idea will spark something though!

saintcrawler, 2017-04-21:
You can write the hostname in a separate file and then execute it from init.lua. Like in [this example](https://nodemcu.readthedocs.io/en/master/en/upload/#initlua) (see the very first lines). Then, you just re-uploads that separate file with nodemcu-uploader.

liv3dn8as, 2017-04-24:
Thanks for the example and yes, that is a possibility but I would still have to have a file for every hostname I wanted to use. My main objective is, for example. Say someone is trying to create a Mesh Network and the configuration files are all the same. Wouldn't it be super simple to just type the wanted hostname while uploading the files to each ESP? I did manage to use 'echo' and append to the init.lua file. Which will have to do for now!