kmpm/nodemcu-uploader #69: nodemcu-uploader running on windows error

JBSchueler, 2017-06-08:
I am trying to run nodemcu-uploader on win10 64bit.
My python installation directory is c:\python27
I used github to get the nodemcu-uploader repository

When I try to run I get the following error
[image]

When I try `setup.py install` it seems ok to me...
[image]

Is it not possible to run nodemcu-uploader on win10?

kmpm, 2017-06-08:
Strange.
I will look into it.
I have been using Win 10 for most of the time without issues.

kmpm, 2017-06-08:
Wrapt seems to be a new unintended dependency.... hmm...

kmpm, 2017-06-13:
Fixed in version 0.4.3.
Closed by d661fac49727825832818754e9779c5bad26f6d9

closed 2017-06-13 by kmpm