kmpm/nodemcu-uploader #70: DeviceNotFoundException: Device not found or wrong port

tulians, 2017-06-13:
Hello, thanks for the tool! I'm still facing the same issue that [happened last year](https://github.com/kmpm/nodemcu-uploader/issues/60). When I try to run `nodemcu-uploader upload init.lua` I get the `DeviceNotFoundException`:

opening port /dev/tty.SLAB_USBtoUART with 115200 baud
Traceback (most recent call last):
  File "/Users/tulians/Documents/learner/learner/bin/nodemcu-uploader", line 11, in <module>
    sys.exit(main_func())
  File "/Users/tulians/Documents/learner/learner/lib/python2.7/site-packages/nodemcu_uploader/main.py", line 252, in main_func
    uploader = Uploader(args.port, args.baud, start_baud=args.start_baud)
  File "/Users/tulians/Documents/learner/learner/lib/python2.7/site-packages/nodemcu_uploader/uploader.py", line 78, in __init__
    __sync()
  File "/Users/tulians/Documents/learner/learner/lib/python2.7/site-packages/nodemcu_uploader/uploader.py", line 76, in __sync
    raise DeviceNotFoundException('Device not found or wrong port')
nodemcu_uploader.exceptions.DeviceNotFoundException: Device not found or wrong port

I tried running `nodemcu-uploader` with different parameters, different start baud rates, using different USB ports,  but still get the same exception. The output of `lsusb` is:

Bus 020 Device 004: ID 0a5c:4500 Broadcom Corp. BRCM20702 Hub 
Bus 020 Device 007: ID 05ac:828f Apple Inc. Bluetooth USB Host Controller 
Bus 020 Device 002: ID 10c4:ea60 Silicon Laboratories, Inc. CP2102 USB to UART Bridge Controller  Serial: 0001
Bus 000 Device 001: ID 1d6b:ILPT Linux Foundation USB 3.0 Bus 

where the device in question is third. I'm using MacOS Sierra.

kmpm, 2017-06-13:
Have you verified that your device actually have a working firmware?

Can you get a working terminal?

If you get a `>` after you connect using `nodemcu-uploader terminal` or `screen /dev/<your port> 115200` or minicom or some other tool described in https://nodemcu.readthedocs.io/en/dev/en/upload/ then you have a working firmware.

Try reset the device as well when connected, just in case it's stuck somewhere and it can not produce the `>` prompt. When resetting you should see something like this..
```
NodeMCU custom build by frightanic.com
	branch: master
	commit: c8ac5cfb912ff206b03dd7c60ffbb2dafb83fe5e
	SSL: false
	modules: bit,file,gpio,i2c,mqtt,net,node,pwm,tmr,u8g,uart,wifi
 build 	built on: 2017-06-01 20:00
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
lua: cannot open init.lua
```

If you don't get any or both of these then you do not have a correct/working firmware on your device and nodemcu-uploader will not be able to help you. The device needs to be reflashed.

kmpm, 2017-06-14:
if you have a recent firmware and a recent version of nodemcu-uploader (>= 0.4.0) then there should not be a baudrate issue.

Sounds as if the firmware in the device is wrong.
Flash again is my recommendation.

tulians, 2017-06-15:
I had to flash the firmware using the guidelines given by [this article](http://www.esp8266.com/viewtopic.php?f=22&t=11431). The board was continuously rebooting. Now nodemcu-uploader works perfectly. Thanks for your help Peter.

closed 2017-06-15 by tulians