kmpm/nodemcu-uploader #72: ESP32 not supported?

joysfera, 2017-06-15:
nodemcu-uploader fails on NodeMCU for ESP32 on my esp32 dev board. No idea why. The 'terminal' works normally, but the 'upload' dies this way:

```
$ python nodemcu-uploader.py -p /dev/ttyUSB0 -t 30 upload init.lua
opening port /dev/ttyUSB0 with 115200 baud
Traceback (most recent call last):
  File "nodemcu-uploader.py", line 10, in <module>
    main.main_func()
  File "/tmp/nodemcu-uploader/nodemcu_uploader/main.py", line 252, in main_func
    uploader = Uploader(args.port, args.baud, start_baud=args.start_baud)
  File "/tmp/nodemcu-uploader/nodemcu_uploader/uploader.py", line 78, in __init__
    __sync()
  File "/tmp/nodemcu-uploader/nodemcu_uploader/uploader.py", line 76, in __sync
    raise DeviceNotFoundException('Device not found or wrong port')
nodemcu_uploader.exceptions.DeviceNotFoundException: Device not found or wrong port
```

```
$ python nodemcu-uploader.py -p /dev/ttyUSB0 -t 30 terminal
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

> 
```

vincegre, 2017-06-25:
I have same problem here :( Found a solution ??

joysfera, 2017-06-25:
Just a work-around. IIRC https://github.com/4refr0nt/luatool worked OK.

kmpm, 2017-07-12:
need to get me a esp32 i think. 
Looks as if the method of detecting a working device is not working as it should.

Gaius-Velleius, 2018-06-01:
I've had the same issue when trying to communicate with the ESP8266 chip on my Adafruit Feather Huzzah. Installing the cp210x.ko kernel module driver from Silicon Labs (which the Adafruit guide links to) didn't fix it either. 

Just like for @joysfera, I can communicate with my device using `nodemcu-uploader terminal ` or `screen /dev/ttyUSB0 9600.` However, I need access to init.lua because `lua: cannot open init.lua`.
 
Whenever I try `nodemcu-uploader download init.lua --port /dev/ttyUSB0` I get 
`nodemcu_uploader.exceptions.DeviceNotFoundException: Device not found or wrong port` as well.

mucek4, 2019-04-19:
As a workaround comment next line in uploader.py
`self.__expect('%sync%\r\n> ')`

joysfera, 2019-04-19:
But NodeMCU on ESP32 does print the '>' char prompt, just like on the ESP8266, doesn't it?
Does it perhaps omit the '\r' or what's up?

mucek4, 2019-04-19:
I don't know, I haven't check. I am not really good at python. I'll try to check.

jasaw, 2019-06-19:
I split the expect sync line to two lines and got me further, but now I'm hitting a different problem.
```python
                self.__expect('%sync%\r\n')
                self.__expect()
```

jasaw, 2019-06-19:
Uploading binary file in verbose mode causes it to print garbage. This patch fixes it. Hope it helps somebody.
```patch
diff --git a/nodemcu_uploader/uploader.py b/nodemcu_uploader/uploader.py
index e12d243..b8a15e2 100644
--- a/nodemcu_uploader/uploader.py
+++ b/nodemcu_uploader/uploader.py
@@ -74,7 +74,8 @@ class Uploader(object):
                 time.sleep(self.autobaud_time) # Wait for autobaud timer to expire
                 self.__exchange(';') # Get a defined state
                 self.__writeln('print("%sync%");')
-                self.__expect('%sync%\r\n> ')
+                self.__expect('%sync%\r\n')
+                self.__expect()
             except CommunicationTimeout:
                 raise DeviceNotFoundException('Device not found or wrong port')
 
@@ -375,7 +376,7 @@ class Uploader(object):
             log.debug('pad with %d characters', padding)
             data = data + (' ' * padding)
         log.debug("packet size %d", len(data))
-        self.__write(data)
+        self.__write(data, True)
         self._port.flush()
         return self.__got_ack()
 
```

joysfera, 2019-09-17:
Tried all above mentioned patches, nothing helped.

joysfera, 2019-09-17:
FYI, working on a real fix. Seems like the issue is related to multiple Lua commands in one line. Doesn't make sense but spliting the commands to separate lines seems to help so I'm giving it a try.

Cyberfonic, 2020-04-30:
I too had this problem, and I started out trying to track the problem down.
Since I'm rather Python v3 focused, I had installed **nodemcu-uploader** with:

    pip3 install nodemcu-uploader

Made a copy of the contents of `~/.local/lib/python3.6/site-packages/nodemcu_uploader`
into an empty directory to start debugging. Immediately had the following error:

    ImportError: No module named serial

Weird? Checked sys.path and it was pointing to python2.7 libraries.

Changed the first line in `nodemcu-uploader.py`

    #!/usr/bin/python3

Re-testing and upload worked.

I guess that was to be expected. Ubuntu 18.04 LTS defaults to Python v2.7.
Wondering whether this single character "fix" might work for others.

Cyberfonic, 2020-04-30:
I'm using a SparkFun ESP-32 Thing and had to rebuild the firmware for 26MHz.

I found that commenting out the auto-baud code in `uploader.py`, i.e.
```
      #self.__writeln('UUUUUUUUUUUU') # Send enough characters for auto-baud
       self.__clear_buffers()
       #time.sleep(0.30) # Wait for autobaud timer to expire
```
helps. BUT not all the time.

When I get the usual error, then I connect as a terminal, enter `=print 'please work'`, exit and try again.  _The actual string printed doesn't matter_.