kmpm/nodemcu-uploader #76: exec executes the file char by char

crasu, 2017-09-28:
Running this:

    #nodemcu-uploader --baud 115200 exec dfplayer-test.lua 

Result in dfplayer-test.lua being executed char by char:

```
opening port /dev/ttyUSB0 with 115200 baud
Execute dfplayer-test.lua
> p
>> =
>> r
> e
>> q
stdin:2: '=' expected near 'q'
> u
...
```

Changing utils.py to this seems to fix it:
```
def from_file(path):
    with open(path, 'rb') as f:
        content = f.readlines()
    return content if PY2 else [ c.decode(ENCODING) for c in content ]
```

kmpm, 2018-12-11:
On which OS are you running this? 
There is a chance of this being different End-Of-Line markers between reality and expected.

crasu, 2018-12-11:
Linux - ubuntu

rvalle, 2020-03-21:
same thing here on linux-fedora
@kmpm char by char should not be EOL related.
anyway, I tested converting it to dos format before submission and fails in the same way.

could it be a serial buffering thing? I am going to test a little, see if I can find out anything...

rvalle, 2020-03-21:
ok I suspect the serial ...

I tried setup virtual environments to see if that would make it work in any conf

```
python 2.7 and:
Package          Version
---------------- -------
nodemcu-uploader 0.4.3  
pip              20.0.2 
pyserial         3.4    
setuptools       44.1.0 
wheel            0.34.2 
wrapt            1.12.1 
```

Fails the same and
```
Python 3.7 
Package          Version
---------------- -------
nodemcu-uploader 0.4.3  
pip              20.0.2 
pyserial         3.4    
setuptools       46.1.1 
wheel            0.34.2 
wrapt            1.12.1 
```
also fails.

do you remember any particular versions that would work on linux?