kmpm/nodemcu-uploader #81: Upload filed for new NodeMCU builds

sleeply4cat, 2018-05-09:
` nodemcu-uploader --port /dev/ttyUSB0 --baud 115200 upload "/home/kirill/111/progr/lua/Washer-small/*"
opening port /dev/ttyUSB0 with 115200 baud
Preparing esp for transfer.
Transferring /home/kirill/111/progr/lua/Washer-small/relay_task.lua as /home/kirill/111/progr/lua/Washer-small/relay_task.lua
Bad chunk response "ANIC: unprotected error in call to Lua API (stdin:6: open a file first)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

����o��;��o|�llld`c�

NodeMCU custom build by frightanic.com
	branch: master
	commit: 5073c199c01d4d7bbbcd0ae1f761ecc4687f7217
	SSL: false
	modules: file,gpio,http,i2c,net,node,spi,tmr,uart,wifi,wps
 build 	built on: 2018-03-25 16:29
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
lua: cannot open init.lua
`

bazooka07, 2018-08-23:
Hello,
in the nodemcu-firmware repository: 
https://github.com/nodemcu/nodemcu-firmware/blob/master/app/include/user_config.h
You can find that :
```C
// maximum length of a filename
#define FS_OBJ_NAME_LEN 31
```
The length of filename can't exceed 31 characters by default.
54 letters are too enough.

sleeply4cat, 2018-08-23:
But file name length is only 12 chars.

bazooka07, 2018-08-23:
Not in Nodemcu. There is not any folder. No pathname.
the filename is "/home/kirill/111/progr/lua/Washer-small/relay_task.lua" without folder separator.
Read the doc :
https://nodemcu.readthedocs.io/en/master/en/modules/file/

sleeply4cat, 2018-08-24:
Omg, I know how SPIFS works.
In this case, why does download tool use a full path as a file name?

kmpm, 2018-12-11:
This issue is related to #24

commit 3d125a82 (2019-12-07): fix: Validate remote path length

Test length of remote path before upload and download.
Closes #81
closed 2019-12-07 by kmpm