kmpm/nodemcu-uploader #90: Call for maintainers

kmpm, 2019-11-13:
Hi,
This project is in need of maintenance and I (kmpm) do not have the time the project deserves.
I really don't use NodeMCU that much any more and am quite busy with other stuff.

If you feel you would like to help out either send me a message or make a comment here.
Would love some help.

TerryE, 2020-05-20:
@kmpm Peter, maybe you could consider adding me as a committer on the project.  You might know that I am one of the main developers on the NodeMCU project, and keep a cut-down version in date for my own use.  Another option would be for me just to update this master.  Up to you.  Regards Terry

kmpm, 2020-05-20:
@TerryE, you should have an invite. 
I have no hesitation adding you.

kmpm, 2020-05-20:
I can publish to pypi, that's not the hard part. But digging up hardware, firmware and triaging bugs is a mess when you have time for it once in a quarter or so. :)

closed 2020-05-20 by kmpm