kmpm/nodemcu-uploader #93: Outdated version in PyPi

galjonsfigur, 2020-08-29:
Latest version of this tool installed via pip from PyPi is 0.4.3 and that version is not working with new NodeMCU firmware. The only workaround is to install it directly from this repository. I know this project is not actively maintained but I think updating it on PyPi would be beneficial and possible without big effort.

kmpm, 2020-09-09:
Hi,
The master branch is the developing branch so it's quite normal that the versions does not match between that and pypi.
There are changes in the master branch that I was waiting for verification on but without no feedback, for example #89

So I thank you because you are the first one confirming that the changes actually work.
I will release a new version in the next couple of days.

galjonsfigur, 2020-09-09:
Sure - if you want any help with testing just ping me. I think this project is really important to NodeMCU firmware and I would be happy to help.

kmpm, 2020-12-02:
Published

closed 2020-12-02 by kmpm